badgerloop-software/hardware #6: Remote IO Initial Schematic Review

rcastle1, 2019-09-08:
Schematic Location: https://github.com/badgerloop-software/hardware/blob/master/braking_io/WorkingSchExports/braking_io_rev4.pdf

Project Location: https://github.com/badgerloop-software/hardware/tree/master/braking_io

Design Files (Power Supply and Bus Filter): https://github.com/badgerloop-software/hardware/tree/master/braking_io/design

ezraboley, 2019-09-09:
It looks like the address pins are left floating on the EEPROM, when they should be either tied high or low according to the schematic

ezraboley, 2019-09-09:
U19 has it's SCL line connected to SDA, and vice-versa

rcastle1, 2019-09-09:
A0, A1 and A2: The Address inputs set the device address
that must be matched by the corresponding Slave address
bits. The Address inputs are hard−wired HIGH or LOW
allowing for up to eight devices to be used (cascaded) on the
same bus. When left floating, these pins are pulled LOW
internally.

If R136/137/138 are populated, they are pulled up to VCC, otherwise if DNP'ed the address pins are pulled low internally according to the datasheet. 
[image]

Also added a bypass cap to U25

Updated SCL and SDA (Will push after all review changes are made)

rcastle1, 2019-09-15:
Notes from design review with Professor Krachey and my thoughts on what still needs to be done

* ESD diodes: maximum rated voltage not present on datasheet. make sure it's set for 20,000V at least. 
* ESD diodes: make sure clamping voltage is less than maximum ADC pin voltage
* Double check all ESD diodes rating. Maybe switch a unidirectional for better negative voltage performance,as these are only rated to -0.3V

* Double check all logic levels. Check that I2C and GPIO levels are appropriate. 

* Update MCU on schematic

*Add 24V reference supply for pressure sensors

rcastle1, 2019-11-04:
done (Rev A)

closed 2019-11-04 by rcastle1