badgerloop-software/hardware #8: ESD Diode Selection

rcastle1, 2019-09-15:
The ESD diodes currently in the schematic were selected as placeholder components. This task is to select appropriate ESD protection diodes for all components of the circuit that interface with the outside world. 

# Motivation

From TI: System-level protection for electrostatic discharge (ESD) is crucial in today’s world, not only in the industrial space but also in the
consumer and automotive space. It only takes one ESD strike to permanently damage a product, which makes ESD protection a critical component in a system design.

# What is ESD? 

Again, from TI: 
[image]

# Definitions
[image]

# What do we need? 
* Fast response time
* Low clamping voltage less than the absolute maximum voltage of the input device
* Operating voltage of the ESD diode higher than the expected operating voltage of the circuit
* Sufficiently high maximum operating voltage of the ESD Diode
* Low capacitance
* Capacity to handle high-voltage ESD strikes
* Be able to handle multiple, repeated ESD strikes

# Different Circuits
- [ ] Pressure Sensors
- [ ] UART
- [ ] Harness Detection circuit
- [ ] Thermistors
- [ ] Heartbeat

# Next Steps
1. Pick a circuit
2. Determine the operating conditions for that circuit from the schematic, latest version here: https://github.com/badgerloop-software/hardware/blob/master/braking_io/WorkingSchExports/braking_io.pdf
3. Spec an ESD diode. Pro-tip, use the parametric search feature on Digikey. Ideally it would have the same footprint as https://www.digikey.com/product-detail/en/littelfuse-inc/SMBJ15A/SMBJ15ALFCT-ND/285981 and possibly even be this same package. A starting place would be to look at the SMBJ15ALFCT datasheet and see if any of those alternate part numbers meet our requirements
4. Get a digikey link to the selected component and comment in this issue.

rcastle1, 2019-10-08:
Selected SMBJ5.0A https://www.digikey.com/product-detail/en/littelfuse-inc/SMBJ5.0A/SMBJ5.0ALFCT-ND/285951 @NerfJohn  can you post a summary of your findings if you get a chance

closed 2019-10-08 by rcastle1

NerfJohn, 2019-10-08:
Looking over the five sensors' datasheets, the lowest (and therefor limiting) voltage seems to be ~5.3V. The lowest breakdown voltage reached by an ESD Diode from the same package as the current one (and basically any other ones found on Digikey) is 6.4V. It's not ideal, but should minimize circuitry damage much better than the current model (which has a low breakdown voltage of 16.7V).