badgerloop-software/hardware #13: Shutdown circuit

dawiggleman, 2019-10-09:
Problem:
The opto-isolator that was turned on to allow for 12V to power the contactor inside the charging box could not source enough current to drive the relay. Instead, we took the output of the shutdown circuit and made a harness to drive the relay if the shutdown circuit was completed.

Solution:
Add an additional net off of the shutdown circuit output to an on-board connector that drive the charge contactor without having to remove the main contactor box drive harness.

Problem:
In order to close the shutdown circuit without controls intervention, 5V needed to be manually jumped with a wire to act as "MCU latch" on the shutdown circuit. Not ideal.

Solution:
Add a simple push-button that when pushed, connects 5V with MCU latch net.

Let's remember to add multipurpose inputs/outputs from BMS in main BMS connector- output can be configured to be activated if temperature goes above a certain temperature which would be nice to have in addition to discharge enable signal

it would be good to have the option for these config inputs to be software controlled in addition to manual switches. this is the same for the charge/discharge/balance enable switches as well.

Ideally, the full pod would be on an monitoring telemetry during charging, but in California it's likely that charging will occur late at night and the software folks should use the sleep.

dawiggleman, 2019-10-11:
Also:
-Add button that toggles charging input

rcastle1, 2019-11-06:
Remove pressure switch

dawiggleman, 2019-11-06:
Remove LV batt net and 1M resistor on BMS charge/discharge section

closed 2019-12-01 by dawiggleman