badgerloop-software/hardware #20: Precharge layout Initial Review

rcastle1, 2019-11-16:
[image]
* Avoid via in pad. Some manufacturers will charge more for it and low cost manufacturers will do it but the quality can be poor. 
* I prefer making all silkscreen "TrueType" rather than stroke, Arial. Reduces silkscreen size and avoid some issues that come up when you have really small silk features. 
* While we're on the topic of silkscreen, feel free to copy/past the bloop and Altium logo from the braking IO board. 
* It's also good practice with HV/LV boards to have clear separation. A line in silkscreen can be handy for this. 
[image]
* A general improvement to the mounting hole footprint would be to add a void around the shape. When we have planes on the top layer, it's possible for the soldermask to degrade and short against the head of the screw if we screw in and unscrew repeatedly. I believe the yellow if the size of the screwhead. 

[image]
* Avoid copper "necks". You can do this with routing or change the copper neck entry thickness in the plane editor

[image]
* More just style, but go directly to the center of pads

[image]

* Avoid this type of double up
* Try to keep three point connections like this 120deg rather than 90/90/180

[image]
* Inductor placement is not good. Keep it as close to U1 pins 5  6 7 8 as possible. This is a large "hot loop" that can cause noise issues to come up, as we have undamped high frequency switching across a large section of the board. 
* You'll want larger floods between OUT+ and the inductor pads. Even though we're not drawing a lot of current, it can cause feedback issues if trace resistance is too high. 

[image]
* I trust you did the math, but 1K seems high for red LEDs

[image]
* For harnessing simplicity, put connectors on the same edge 
* Adding labels in silkscreen is also good practice. 

[image]
* C1 placement could also be improved. Think about what it needs to connect to (on both sides) and keep it short. 
* Not sure if I have the latest version, but is that fuse footprint correct? 23

Random thoughts:
* Seems like P1s connections all go to the other side of U1. Consider moving that over. 
* Overall, I think we need to more clearly think about keeping HV and LV separate, and avoiding crossovers whenever possible..
* Mounting holes are kinda in weird spots. 
* Could probably condense the board a bit by offsetting the precharge resistors. not sure how worth it that would be though.

dawiggleman, 2019-11-17:
Thanks for the review, great points. Regarding vias in planes, would you recommend the thermal relief connection, or just having a single trace go to the pad? @rcastle1

dawiggleman, 2019-11-17:
Also, are there general rules of thumb on what to have pours of vs individual traces?

rcastle1, 2019-11-17:
Via in plane should be direct connect. Thermal reliefs are used so that you can solder a component down more easily, and you shouldn't have to solder anything to a via. 

Use pours when you have high current, when you need a low resistance / low impedance path, or for  ground floods to minimize return paths. Polygons/planes are common for power and GND. That being said, copper is copper, so if you have the same amount of copper it doesn't matter if it's a plane or a trace. 

Here's some more info: https://www.altium.com/live-conference/theme/PDFs/SDay%201/Dan_Beeker_KEYNOTE_EM%20Fields%20for%20Normal%20Folks.pdf

rcastle1, 2019-11-17:
https://www.youtube.com/watch?v=WglPHiZx4Gw&feature=youtu.be

dawiggleman, 2019-11-18:
Regarding "avoiding via in pad", that component is through hole, so those are the holes are for the component itself and the pour goes around it. Were you thinking it was a surface mount component?

rcastle1, 2019-11-18:
Gotcha. Yeah, it looked like it was SMD given the flood around it. looks fine.

closed 2020-01-12 by dawiggleman