badgerloop-software/hardware #25: TPS54531 Library Component Request

rcastle1, 2019-11-21:
Hello @dawiggleman (current librarian), please add the following to the ic.SchLib and ic.PcbLib libraries:

* TPS54560BQDDARQ1 
    * Datasheet link: http://www.ti.com/general/docs/suppproductinfo.tsp?distId=10&gotoUrl=http%3A%2F%2Fwww.ti.com%2Flit%2Fgpn%2Ftps54560b-q1
    * Digikey Link: https://www.digikey.com/product-detail/en/texas-instruments/TPS54560BQDDARQ1/296-45534-1-ND/6823164
    * Footprint: HSOP (8)
    * Notes: Includes a thermal pad at the bottom

* SRR1280-4R7Y
    * Datasheet link: https://www.bourns.com/docs/Product-Datasheets/SRR1280.pdf
    * Digikey link: https://www.digikey.com/product-detail/en/bourns-inc/SRR1280-4R7Y/SRR1280-4R7YCT-ND/2127446
    * Footprint: 0.492" L x 0.492" W (12.50mm x 12.50mm)

Initial usage: 12V Power Supply for bus to motor controller power conversion.

dawiggleman, 2019-11-21:
Do you want the inductor in the inductor library? Was also considering the inductors that are in misc library over to inductor library... would that cause issues downstream to the braking io board?

rcastle1, 2019-11-21:
Yeah inductor library is fine. I'd prefer to keep it split up so we can easily check through the existing inductor library to encourage part reuse, but not a big deal. For the braking IO board, I save a zip file of the project and the libraries at time of release, so we can recreate the project at time of release. Otherwise, it's fine to just update the component references if components get moved around.

dawiggleman, 2019-11-22:
So to confirm, it would be okay to move the inductors in the misc library to the inductor library?

closed 2019-12-01 by dawiggleman