badgerloop-software/hardware #32: New Part Request

sluggercook, 2019-12-06:
Digi-Key Part Number | 732-61802525023-ND

Digikey
https://www.digikey.com/product-detail/en/61802525023/732-61802525023-ND/10484647?utm_campaign=buynow&WT.z_cid=ref_octopart_dkc_buynow&utm_medium=aggregator&curr=usd&site=us&utm_source=octopart

Datasheet
https://www.we-online.de/katalog/datasheet/6180xx25023.pdf

rcastle1, 2019-12-09:
Would one of these work for you instead @sluggercook ? We ran into a lot of issues with DSUBs last year. (Solder connection, screwing/unscrewing the backshell, locking feature not being a part of the mating process)

https://www.molex.com/molex/products/datasheet.jsp?part=active/0430452400_PCB_HEADERS.xml

rcastle1, 2019-12-09:
Also this might go in a separate thread, but I'd be curious what the plan is for voltage taps and thermistors for the pack.

sluggercook, 2019-12-09:
@rcastle1 
So I was planning on using the DSUB connector on the inside of the breakout board (the battery side) and a right angle Molex on the outside of the board, so if we were going to use a Molex connector on the inside, it would have to be a straight connection, not right angle.  The only issue I thought might arise with the Molex connectors on the inside was them possibly sticking out too far over the batteries inside of the pack, making it harder to pull the batteries in and out easily.  I couldn't find any actual dimensions but just based off pictures it looked like they might stick out too far.  If you don't think it would be a problem though, we can do a straight connection Molex.  As far as the voltage taps and thermistors go, the plan with the 24 pins per pack were to have 15 voltage taps (probably secured on the batteries with ring terminals), 7 thermistors, and 2 grounds.

dawiggleman, 2019-12-09:
Thinking about it, I think it might be worth looking at the molex model in CAD before reverting back to Dsubs. The vertical mount one wouldn't be massively bigger than the JST connectors from last year, and if we need to take the cells out we could unplug the connector

dawiggleman, 2019-12-09:
https://www.digikey.com/products/en?keywords=%20430452412

dawiggleman, 2019-12-09:
This vertical mount would stand off board 9.91 mm, whereas the JST from last year (https://www.digikey.com/product-detail/en/jst-sales-america-inc/B4B-XH-A-LF-SN/455-2249-ND/1651047) stands off 9.8 mm, so pretty much the same

sluggercook, 2019-12-09:
Okay then we'll go with that vertical mount Molex

closed 2020-01-17 by dawiggleman